efabless/caravel-gf180mcu #31: simple_por power pins

mo-hosni, 2022-11-18:
simple_por power has a mismatch in power pins between the RTL and the LEF/GDS views and the power pins are not easily accessible.

Suggestions:
- Power pins be VDD and VSS in all views.
- Power rings to be added.

RTimothyEdwards, 2022-11-21:
@mo-hosni :  I need to know what size/pitch/layers to make the power rings.

mo-hosni, 2022-11-22:
@RTimothyEdwards: Well, I locally changed the pin names in the LEF view from `vdd` and `vss` to `VDD` and `VSS` to match the RTL, and I was able to connect the por to the PDN using a custom script. So, I think we can close this issue and then review the power connections once I make a pull request with the full caravel.

closed 2022-11-24 by mo-hosni