efabless/caravel-gf180mcu #46: Simple por is not LVS clean

mo-hosni, 2022-11-23:
Simple por has a short between vdd and vss.

I found that when running LVS at caravel_core. Then, I traced the nets in por and found that vdd and vss are shorted. Also, it is not DRC clean.

RTimothyEdwards, 2022-11-23:
@mo-hosni :  This appears to be an issue with reading GDS into magic.  The layout looks correct, and extracting and running LVS from the magic view is correct.  The only thing that is screwed up is reading GDS into magic and then extracting and running LVS.  It should require only a quick fix to the magic tech file.

mo-hosni, 2022-11-23:
@RTimothyEdwards: Maybe, but I used Klayout not Magic.

I read the por GDS using klayout. Ran a hierarchical trace for all nets. Then found from Klayout that vdd and vss are the same net.

RTimothyEdwards, 2022-11-23:
@mo-hosni :  The klayout tech file is wrong, then.  There is no short between `vdd` and `vss`.  Where is it identifying one?

RTimothyEdwards, 2022-11-23:
@mo-hosni :  So this appears to be three separate issues.  I have fixed the issue with GDS read-in that I found, which may or may not be related to either the DRC or LVS errors, depending on how those were generated.  If they came from magic after reading in GDS, then that is the most likely cause.  But it appears that the klayout deck is incorrect, and the layout itself still has the issue with the lower-case names for the power supply.

I'm going to assume that the layout is fundamentally correct and just make the minimal changes necessary to make the power supply names uppercase.

RTimothyEdwards, 2022-11-23:
Partially solved by pull request #47 (makes the power supplies upper-case in all views).

closed 2022-11-23 by jeffdi