efabless/caravel-gf180mcu #63: gen_gpio_defaults.py need to be updated for gf180 and the new caravel RTL updates

M0stafaRady, 2022-11-29:


RTimothyEdwards, 2022-11-29:
What needs to be fixed is the gate level verilog netlist for the gpio_defaults_block module.  This needs to be in a very specific format, following what was done with the Sky130 versions of the cells, which is a netlist with all of the components but with pins connections made by a series of assignment statements at the end of the cell.  The script expects to see this format or else the script will not work.

RTimothyEdwards, 2022-12-01:
Gate level verilog is fixed by #68 ;  need to check that the scripts work with the modified netlists.

RTimothyEdwards, 2022-12-01:
More changes were made with an additional commit to PR #68.

Note that PR #68 changes both the gate level netlist and the script.  Kareem's other pull request (#66) accomplishes the same thing (with automation, which is good) but there will probably be issues with missing decap cells, and the power pins will need to have VNW and VPW added when the PDK is updated.