efabless/caravel-gf180mcu #69: Simple POR is not DRC clean

marwaneltoukhy, 2022-12-01:
## Expected Behavior
`simple_por` DRC clean

## Actual Behavior
`simple_por` has a violation in rule `LPW.2a_5V_`

## Steps to Reproduce the Problem

1. build the pdk
2. run `python3 run_drc.py --gf180mcu=C --path=$CARAVEL_ROOT/macros/simple_por/gds/simple_por.gds.gz`

## Specifications

- Klayout Version: 0.28
- Platform: Debian

## Extra materials

Screenshot of DRC error:

[image]

marwaneltoukhy, 2022-12-01:
Looks like the DRC rule is wrong, there is no `DNWELL` inside the `simple_por` at all. I can see from the rule that there is `.overlapping(dualgate)` not `dnwell`. @atorkmabrains can you please weigh in here?

atorkmabrains, 2022-12-01:
@marwaneltoukhy May you elaborate why you are using LVPWELL this way? It's a DRC issue as well. But I think it's useless to add this LVPWELL. I just want to understand the reason.

RTimothyEdwards, 2022-12-01:
@atorkmabrains :  I seem to recall having this conversation before.  The GF documentation is very clear where it says "If this layer is used without DNWELL, the body of all those transistors will by default be connected to p-substrate potential", indicating that it is a standard use case to use LVPWELL as an indicator of p-substrate outside of deep nwell.  All of the GDS files from GF use the LVPWELL layer in this way.  This usage also has the advantage that a cell layout, such as a standard cell layout, can be placed directly inside a deep nwell structure and work correctly without modification.  That is, when placed inside the deep nwell area, the LVPWELL indicates a pwell-within-deep-nwell, while when placed outside the deep nwell area, the LVPWLL indicates the p-substrate.  This works without the need for a separate set of standard cells to be used in deep nwell.

RTimothyEdwards, 2022-12-01:
@atorkmabrains :  Note that when used outside of deep nwell, and as an indicator of p-substrate, there is no physical mask geometry associated with the LVPWELL layer and therefore no specific width or spacing rules apply.

atorkmabrains, 2022-12-01:
Thanks @RTimothyEdwards I know that and I wanted to confirm the usage methodology. @marwaneltoukhy you are right. that's clearly a bug in the DRC rule deck. Will be updated. I'm going to release a major update.

marwaneltoukhy, 2022-12-02:
@atorkmabrains We need that change ASAP, as it is blocking our tapeout. When can we get that fix?

atorkmabrains, 2022-12-04:
HI @marwaneltoukhy and @RTimothyEdwards 

All rules has been fixed for simple_por. Please find the latest rule deck at:
https://github.com/efabless/globalfoundries-pdk-libs-gf180mcu_fd_pr

Run command has changed to:
```bash
python run_drc.py --variant=C --path=./testing/testcases/spm.gds --run_dir=./drc_runs
```

Please let me know if you have any other issues.