efabless/caravel-gf180mcu #79: simple_por is not LVS clean

marwaneltoukhy, 2022-12-05:
simple_por is not LVS clea, maybe it needs to be redone using the new PDK? Attached the output of netgen.
[simple_por.txt](https://github.com/efabless/caravel-gf180mcu/files/10155514/simple_por.txt)

RTimothyEdwards, 2022-12-05:
Magic GDS import error.  It works correctly when extracted from .mag, and the GDS is correct.  The MiM cap contacts are getting lost when reading the GDS back into magic.  I will fix it.

RTimothyEdwards, 2022-12-18:
This was fixed in magic for the LVS flow through netgen;  however, the `simple_por` circuit has remained not LVS clean when using Calibre.  I tracked this down to a requirement for the capacitor length marker and added it to the `simple_por` GDS file in pull request #168.

@marwaneltoukhy :  Please check that the `simple_por` can be checked to the transistor level after merging #168 and is LVS clean in Calibre.  If so, this ticket can be closed.

closed 2022-12-18 by jeffdi