efabless/caravel-gf180mcu #87: SRAM is not LVS clean

marwaneltoukhy, 2022-12-06:
SRAM is only LVS clean when doing an abstract LVS, when doing transistor level LVS is is not clean. I attached the report.
[gf180_ram_512x8_wrapper-gds-vs-v.txt](https://github.com/efabless/caravel-gf180mcu/files/10166996/gf180_ram_512x8_wrapper-gds-vs-v.txt)

netgen LVS script:
```
set circuit1 [readnet spice /home/marwan/utilities/LVS/spice/gf180_ram_512x8_wrapper-gds-extracted.spice]
set circuit2 [readnet spice /home/marwan/pdks/gf180mcuC/libs.ref/gf180mcu_osu_sc_9T/spice/gf180mcu_osu_sc_9T.spice]
readnet spice /home/marwan/pdks/gf180mcuC/libs.ref/gf180mcu_fd_io/spice/gf180mcu_fd_io.spice $circuit2
readnet spice /home/marwan/pdks/gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/spice/gf180mcu_fd_sc_mcu7t5v0.spice $circuit2
readnet spice /home/marwan/pdks/gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu9t5v0/spice/gf180mcu_fd_sc_mcu9t5v0.spice $circuit2
readnet spice /home/marwan/pdks/gf180mcuC/libs.ref/gf180mcu_fd_ip_sram/spice/gf180mcu_fd_ip_sram__sram64x8m8wm1.spice $circuit2
readnet spice /home/marwan/pdks/gf180mcuC/libs.ref/gf180mcu_fd_ip_sram/spice/gf180mcu_fd_ip_sram__sram128x8m8wm1.spice $circuit2
readnet spice /home/marwan/pdks/gf180mcuC/libs.ref/gf180mcu_fd_ip_sram/spice/gf180mcu_fd_ip_sram__sram512x8m8wm1.spice $circuit2
readnet spice /home/marwan/pdks/gf180mcuC/libs.ref/gf180mcu_fd_ip_sram/spice/gf180mcu_fd_ip_sram__sram256x8m8wm1.spice $circuit2
readnet spice /home/marwan/pdks/gf180mcuC/libs.ref/gf180mcu_osu_sc_12T/spice/gf180mcu_osu_sc_12T.spice $circuit2
readnet verilog /home/marwan/caravel_mgmt_soc_gf180mcu/verilog/gl/gf180_ram_512x8_wrapper.v $circuit2
lvs "$circuit1 gf180_ram_512x8_wrapper" "$circuit2 gf180_ram_512x8_wrapper" /home/marwan/pdks/gf180mcuC/libs.tech/netgen/gf180mcuC_setup.tcl /home/marwan/utilities/LVS/spice/gf180_ram_512x8_wrapper-gds-vs-v.out -json
```
magic extraction script:
```
set extdir $::env(ext_out)/tmp_ext
file mkdir $extdir
cd $extdir
crashbackups stop
drc off
gds flatten true
gds noduplicates true
gds read $::env(ext_inp1)
load [file rootname [file tail $::env(ext_inp1)]] -dereference
select top cell
expand
extract do local
extract no all
extract unique
extract all
extract
ext2spice lvs
ext2spice -o $::env(ext_out)/[file rootname [file tail $::env(ext_inp1)]]-gds-extracted.spice
exit
```

RTimothyEdwards, 2022-12-06:
@marwaneltoukhy :  While I'm working on it, I would like to point out that (1) `-dereference` does absolutely nothing when loading a cell read from GDS, and (2) `extract all` followed by `extract` runs the extraction twice.

RTimothyEdwards, 2022-12-06:
@marwaneltoukhy :  The main issue here is that you are reading a poorly designed vendor GDS file directly into magic.  Use the script in open_pdks in front of the `gds read` command when you are doing the extraction.  The script can be found in open_pdks under `gf180mcu/custom/scripts/gds_import_sram.tcl`.

RTimothyEdwards, 2022-12-06:
@marwaneltoukhy :  And what is the point of reading every single GF library in order to run LVS on a self-contained SRAM?

marwaneltoukhy, 2022-12-06:
> @marwaneltoukhy : And what is the point of reading every single GF library in order to run LVS on a self-contained SRAM?

This is a generic script, I don't think there's harm in reading all the library spice, is there?

d-m-bailey, 2022-12-06:
@marwaneltoukhy @RTimothyEdwards I took a look at the spice file for the sram and it looks like magic extracted data. The cdl file looks like calibre extracted file. Is there no true netlist?
If you’re comparing against extracted data and the extraction program changes, you’re out of luck. The only thing you can do is replace the library version with the extracted version, which doesn’t really check anything, does it?

RTimothyEdwards, 2022-12-06:
@d-m-bailey :  The CDL comes from GF and so is presumably usable as a "golden reference" except for the fact that it is CDL and netgen is very limited in reading CDL syntax.  But I was trying to work on what I thought would be the much simpler problem of (1) making sure all the extracted devices look more or less correct by visual inspection of the netlist (e.g., no 3.3V devices), and (2) be able to run LVS on a netlist extracted from GDS vs. the one that ends up in the PDK. . . which is supposed to be extracted the same way, except that very subtle differences between the way these are extract is resulting in netlists that do not match in netgen.  And the differences are complicated and not simple to analyze.

d-m-bailey, 2022-12-06:
@RTimothyEdwards The CDL has subckts like
```
.SUBCKT ICV_11 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16
** N=16 EP=15 IP=24 FDC=16
*.SEEDPROM
X0 5 6 2 3 4 9 11 10 12 018SRAM_cell1_2x $T=0 0 0 0 $X=-340 $Y=-340
X1 7 8 2 3 4 13 15 14 16 018SRAM_cell1_2x $T=3000 0 0 0 $X=2660 $Y=-340
.ENDS
```
which are calibre created virtual cells that don't correspond to anything in a corresponding netlist. Can anyone reverse engineer the extracted layout into an xschem viewable schematic? That would make more sense as an open source asset,  I think.

RTimothyEdwards, 2022-12-06:
@d-m-bailey :  I assume that's Calibre's way of dealing with the horrible layout hierarchy.  In theory, flattening everything to the transistor level should work, although it's a bit of a pain to set up the GDS read-in script to flatten every cell except for the top level.

RTimothyEdwards, 2022-12-19:
@marwaneltoukhy :  I am now able to get a clean LVS on the SRAM with a combination of methods.  The open_pdks installer has a script that flattens numerous subcells of the SRAM to get a layout that is properly extractable.  Then if I flatten the entire SRAM in magic when I extract the netlist, it will LVS cleanly in netgen.