efabless/caravel-gf180mcu #104: chip_io LVS transistor level failure

marwaneltoukhy, 2022-12-08:
As per the discussion with @d-m-bailey and @RTimothyEdwards, `chip_io` is failing LVS transistor level because of missing spice subcircuits in the PDK

subcircuits missing:
```
gf180mcu_fd_io__bi_t
gf180mcu_fd_io__in_c
gf180mcu_fd_io__dvss
gf180mcu_fd_io__dvdd
gf180mcu_fd_io__in_s
```

marwaneltoukhy, 2022-12-13:
Changes that needed to be done in gl netlist:
1. I had to add `gf180mcu_fd_io__fill5` to the gl netlist as it wasn't there
2. `const_one` bits were reversed

After these changes calibre LVS only has property errors.
To make sure that it is not an issue with `chip_io`, I ran LVS on `gf180mcu_fd_io__bi_t` standalone (cdl from library VS gds from library), which produced property errors.

RTimothyEdwards, 2022-12-13:
@marwaneltoukhy :  Have you committed those corrections and made a pull request?

RTimothyEdwards, 2022-12-13:
@marwaneltoukhy :  The existing verilog/gl/chip_io.v does not have either fill5 or fill10.  To make device counts match, I had to add:
```
  gf180mcu_fd_io__fill10 filler10 [1031:0] (
    .DVDD(vdd),
    .DVSS(vss),
    .VDD(vdd),
    .VSS(vss)
  );
  gf180mcu_fd_io__fill5 filler5 [2:0] (
    .DVDD(vdd),
    .DVSS(vss),
    .VDD(vdd),
    .VSS(vss)
  );
```

marwaneltoukhy, 2022-12-14:
@RTimothyEdwards Calibre only flagged the fill5, and when I tried to add fill10, it flagged device mismatch (more cells in source than layout). Still investigating this behavior.

closed 2022-12-14 by jeffdi