efabless/caravel-gf180mcu #169: tapeout seems to use .mag rather than .gds

gatecat, 2022-12-19:
Previous tapeouts, at least MPW5 on sky130, and also moreover the precheck for gfmpw0 seemed to be running on the .gds view in the submission repo.

To avoid my repo becoming too big and save upload time I only ever committed updated GDS and my generated tapeout caravel gds/oas now seems to contain the example project, from the stale .mag in my repo.

jeffdi, 2023-07-24:
Needs review.

d-mitch-bailey, 2023-08-28:
@gatecat Thanks for reporting. 
There may have been some problems with earlier versions of the `ship` target of `caravel-gf180mcu/Makefile`.

You are correct in that the gds should override the `user_project_wrapper` mag file.
However, it is important to remember that recreating lower levels of the hierarchy will **not** automatically be reflected in the top  `user_project_wrapper`. You must also recreate each parent level gds in order, from lowest to highest, to ensure that your changes appear in the final `user_project_wrapper` gds.

If you have recreated the gds at each level, can you verify that there is still a problem with tapeout using stale mag data?

closed 2024-12-05 by DavidRLindley