efabless/caravel-gf180mcu #176: Need "locking disable" as part of the "make ship" and "make truck" scripts.

RTimothyEdwards, 2023-03-02:
Magic's file locking mechanism is designed for preventing users on a multi-user system (or processes on a multi-process system) from overwriting each other's layouts.  However, it uses the standard file locking mechanism in Linux, which is to leave an open file descriptor to hold the lock on the file.  Linux has a relatively low maximum limit on open file descriptors (< 1024) while the GF I/O library has a relatively large number of cells (> 1024).  This leads to a failure to create new file descriptors, followed by a failure to open (otherwise perfectly good) files, ultimately producing unpredictable results, and bad layout/GDS.  Executing the command "locking disable" at the beginning of a script avoids the problem altogether.

RTimothyEdwards, 2023-07-26:
@jeffdi :  I created a pull request with the correction some time ago, but it has been ignored.

jeffdi, 2023-08-23:
@jeffdi to merge PR