efabless/caravel-gf180mcu #193: gfmpw-1 bidirectional constants not correct. input is disabled.

d-m-bailey, 2023-11-08:


RTimothyEdwards, 2023-11-15:
output enable is not used when oe_override is not set, so it is clear that the 3rd bit should have been set and not the 4th bit:
```
#define GPIO_MODE_MGMT_STD_BIDIRECTIONAL 0x005
#define GPIO_MODE_USER_STD_BIDIRECTIONAL 0x004
```

d-m-bailey, 2023-11-15:
@RTimothyEdwards I only added the `INPUT_ENABLE` bit. I did not change the `OUTPUT_ENABLE` bit. My understanding was that `OUTPUT_ENABLE` was irrelevant.
```
#define GPIO_MODE_MGMT_STD_BIDIRECTIONAL 0x00d
#define GPIO_MODE_USER_STD_BIDIRECTIONAL 0x00c
```
Is this a problem?

RTimothyEdwards, 2023-11-15:
@d-m-bailey :  Not a real problem, no, but maybe confusing to the end-user.  I interpreted the original problem as just being an accidental bit shift, so fixed it accordingly.  But what we really need is just to have a truth table somewhere to properly document the function, presumably in comment lines in the `defs.h` file:
```
OE  OE_override  function
--------------------------------------------------------------------------------------------------------
0     1                     output (forced) disabled
1     1                     output (forced) enabled
X     0                     output determined by OE input, not OE configuration bit
```